robotics-in-concert/rocon_qt_gui #184: [rqt_remocon] ambiguous ROS_HOSTNAME error

stonier, 2015-03-17:
When `ROS_HOSTNAME` is not set, you get an error printed in the shell.

```
[error] Rqt Remocon: 'ROS_HOSTNAME' 
[debug] Interactive Client : initialised
[debug] Interactive Client : Connection Details
```

1. The gui doesn't abort
2. This is not actually an error if you are connecting over localhost

To fix:

Just move it to a warning. Would be nice if the gui had some way of showing what the `ROS_HOSTNAME`/`ROS_IP` settings are.

dwlee, 2015-03-18:
* I will change the error into warning if user does not set ```ROS_HOSTNAME```. In this case rqt remocon use ```localhost``` as ```ROS_HOSTNAME```

* I think that it is good way to use widget object name to show ```ROS_HOSTNAME/ROS IP```. For example, 
```
Role Chooser : ROS IP [http://localhost:11311  / ROS HOST NAME [localhost]
```

stonier, 2015-03-20:
I think you mean: 

```
Role Chooser - ROS_MASTER_URI [http://localhost:11311]  / ROS_HOST NAME [localhost]
```